OpenLightingProject/rp2040-dmxsun #5: Implement automatic binary generation using GitHub Actions

kripton, 2021-02-22:
The binaries for the Pico board should easily be generated by some kind of CI pipeline

peternewman, 2021-02-22:
Happy to help with this as I've done various similar bits in Actions but doing a GitHub Pages thing to publish may be easier (as the actions artefacts expire after a while and only appear as zips).

I'm assuming RPF have some pretty good docs on it, I guess someone might have made an Action already!

kripton, 2021-02-23:
I am new to GitHub Actions and I was honestly curious to try them out finally :) Didn't find anything on the marketplace and also a quick internet research didn't bring up anything meaningful. Thus, I decided to base off the `cmake` pipeline, add the required steps to set up the Pico toolchain and SDK and off it goes :)
I completely agree that gh-pages might be preferable and now we do have a prebuilt binary in the `bin` folder and the ones generated by Actions from the source. Don't know what's easier for users to find.

closed 2021-02-23 by kripton

peternewman, 2021-02-23:
> I am new to GitHub Actions and I was honestly curious to try them out finally :)

You beat me to it! I got as far as opening a new file on my lunch break then got distracted.

> I completely agree that gh-pages might be preferable and now we do have a prebuilt binary in the `bin` folder and the ones generated by Actions from the source. Don't know what's easier for users to find.

My example may not be hugely helpful as it's somewhat niche, but this uses GH pages (in this case to host a deb):
https://github.com/peternewman/Date-Holidays-GB/compare/actions...peternewman:actions-gh-pages-apt

If you're happy to have the noise in your repo, branches are really easy, there's a commit action. You could link to gh-pages from the readme.

Things to consider:
branch builds - do-able with pages, would just need a bit of thought as to how they're listed to make multiple ones available
PR builds - probably want to be actions only
I think commit scope would be similar in both cases.

PR #8 (2021-02-23): Gh actions
Implement GitHub Actions to automatically generate the .uf2-file from the source. Fixes #5