OpenLightingProject/rp2040-dmxsun #6: Implement a "switch to bootloader" command

kripton, 2021-02-22:
When flashing the Pico over USB (instead of using the SWD-based debug port and OpenOCD), one needs to replug the board everytime it should be flashed. If fhe SDK allows it, add a command to the USB protocol to reboot to the bootloader.

2bndy5, 2021-04-03:
just passing this along:

>If fhe SDK allows it, add a command to the USB protocol to reboot to the bootloader.

It would seem that it does support this. I haven't found any example yet, but I found [_**this macro**_](https://github.com/raspberrypi/pico-sdk/blob/1769968c426555454ab88917115abcb96bcfd1cd/src/rp2_common/pico_stdio_usb/include/pico/stdio_usb.h#L54) in search of how to redirect UART I/O to USB (or vice versa). I imagine that using this defined baudrate is how the [unofficial ArduinoCore-pico](https://github.com/earlephilhower/arduino-pico) is automatically resetting the board when re-programming the pico (or other RP2040 based boards).

There's also [this library](https://datasheets.raspberrypi.org/pico/raspberry-pi-pico-c-sdk.pdf#%5B%7B%22num%22%3A230%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C115%2C377.984%2Cnull%5D) which seems like it would be commonly used by CircuitPython/MicroPython firmware to let the user reset-to-bootloader when the reset button is double clicked.

kripton, 2021-04-20:
Fixed with https://github.com/OpenLightingProject/rp2040-dongle/commit/dd2dc2cf51eed4e9961ec046b7b0a472e5cc146c
@2bndy5: This seems to be some kind of "official" way to do it. It can be enabled in the pico-sdk by adding some #defines to a project, assume it's using the stdio_usb interface. See https://github.com/raspberrypi/pico-sdk/commit/383e88ea165bc63e5755e042608bbc1164e75ab7

closed 2021-04-20 by kripton

kripton, 2022-01-08:
Reboot-to-Bootloader is now also available via the HTTP API (integrated webserver): 
https://github.com/OpenLightingProject/rp2040-dongle/blob/79556f4ebf31f9bb41bdc5a23bae965987fad62d/src/webserver.cpp#L31