OpenLightingProject/rp2040-dmxsun #15: Simplify Logging Code

peternewman, 2021-03-01:
You've currently got e.g.
https://github.com/OpenLightingProject/rp2040-dongle/blob/08e6e144c32b5bf002fd0e936cf510dbeb2a6162/src/jaruleproto.c#L15

In OLA we use `__FILE__` too, but we also setup a macro to centralise and tidy most of the back-end commands (although I guess the latter might be a bit harder if it's all in a printf):
https://github.com/OpenLightingProject/ola/blob/ea3a18a925f123c8771ebc6c60327678e99ddcf2/include/ola/Logging.h#L59-L60

kripton, 2021-03-01:
Indeed, fully agree. That is meant to be a macro eventually. For the moment, it's just written to fulfil the needs.
I did try `__LINE__`, however it contains the full file path and that was a bit long for me. I saw some functions online that return only he filename but I didn't yet bother to integrate them.

peternewman, 2021-03-03:
> I did try `__LINE__`, however it contains the full file path and that was a bit long for me. I saw some functions online that return only he filename but I didn't yet bother to integrate them.

I assume you mean `__FILE__`? That's interesting, in OLA we just get the relative path (say `src/jaruleproto.c`).

kripton, 2021-03-03:
Indeed, I meant `__FILE__`. Maybe it depends on the build system used? It could be that OLA's GNU Make-based build system calls the compiler with relative paths while the CMake-based build system for Pico-projects calls the compiler with absolute paths?

kripton, 2021-03-04:
I played around a little with the logging code. Actually, rather for checking if C++ can be used on the RP2040. Result: Working, but some things (such as iterators) will lead to compile-time errors such as https://github.com/purduesigbots/pros/issues/134

Current status at https://github.com/OpenLightingProject/rp2040-dongle/tree/kripton-cppTest. Can be used as a base but I might rewrite that in plain C. Will see ;)

closed 2021-12-22 by kripton