OpenLightingProject/rp2040-dmxsun #17: Any harware refrance.

ScobbyDoo, 2021-12-29:
Hi, 
Im looking for any schematic to make one & try 
If you have schematic i wll design pcb & share it.

kripton, 2021-12-29:
Hi @ScobbyDoo and I'm happy that you want to try it.
The schematic and PCD design files are in the "hardware" branch: https://github.com/OpenLightingProject/rp2040-dongle/tree/hardware. The files are created using KiCad.
It's basically the baseboard + some baseboard options that you can connect to the base board. I have several ones here, ordered at JLCPCB.

Of course, you can also do a prototyping solution and simply hook up some RS-485-Driver breakout-boards to the RPi Pico board via jumper wires. The connections can be seen in the KiCad-schematics but don't hesitate to ask here if something is unclear.

Each IO board usually contains one I2C-EEPROM to give the base board some information about the IO board (available ports, port direction, .....) but at the moment, the firmware just assumes 16 OUTs

kripton, 2021-12-29:
Oh, and the KiCad-files are automatically rendered to PDFs using GitHub actions. The latest results are here: https://github.com/OpenLightingProject/rp2040-dongle/suites/4741501920/artifacts/130997826

ScobbyDoo, 2022-01-03:
But hiw do i send command to drive dmx?

Is it standalone system ,or usb , Ethernet option available.

kripton, 2022-01-03:
Since I don't know your skill level, regarding DMX-512 hard- and software, it's a bit hard to guess where to start.
If you flash the firmware to an RPi Pico board, it will reboot and start generating valid DMX data right away. However, it's all ZEROs of course. To get it to send meaningful data, you need to connect the Pico-board to some kind of PC via USB. If the firmware is running properly, the Pico board will look to the PC like an Ethernet network card. So it's not real Ethernet but your PC will see a new network card (if your OS is not too old) and will get an IP address from the Pico board via the DHCP protocol. The exact IP that you need to talk to the board differs from board to board (so we can support multiple boards connected to the same PC) but will be in the range of 169.254.XXX.1. You can see the IP if you run `ip addr` on Linux or if you hover the "USB devices" task bar icon on Windows.

Then, you can fire up your favourite web browser and enter the board's IP in the address bar. The web interface of the board should open, showing you the general system status. At the top, there is a item called "Console". If you click there, you will be able to modify the values via the web browser.

The other alternative is to use some kind of DMX-data-generating software such as OLA or QLC+. You can send E1.31 (sACN) or ArtNet frames from that software running on your PC to the Pico board via the virtual Ethernet connection. The Pico board should react to those values and output the data as DMX-512 on one of the GPIOs (depending on universe value in the sACN or ArtNet frames). You should also see the values change on the web interface.

All this can be done without additional hardware. The hardware is only needed if you want to actually connect DMX-512 fixtures, use the wireless modes or want to see the board's status on the status LEDs.

... and I know that I should update the Readme file with all of this information ;)

kripton, 2022-01-03:
So to answer the questions:
It doesn't have a real hardware Ethernet option as of now. Of course, one could attach an SPI or UART-based Ethernet module but it's not yet supported.
It works when connected to a host (PC or RaspberryPi or similar) via USB.
It can also work stand-alone when powered via micro USB connector (power bank or wall-USB power adapter) and when using the wireless module.

ScobbyDoo, 2022-01-05:
Yes, I'm familiar with artnet & e1.31

I will attach Ethernet to controller & send data over udp to mcu ,as per calculation (16 x 512 byte ) universes how do i assign those data to specific data lines 

Like store first 512 byte to 1st data line 
2nd 512 to 2nd data line 
& So on , fill full buffer & driver it on rs485 line 

Now the point is how do i fill that buffer & drive it? 
I suppose memcpy function may works for that😅

Sorry for silly logics(if their is any)

kripton, 2022-01-07:
Where to you plan to attach the LAN8720a? As I wrote above, the firmware won't recognize or use it. The Ethernet-Emulation uses the USB connection to the host PC. You don't need an Ethernet-cable if you want to send data from your PC to the dongle.

The assignment of the data to the "data lines" is done automatically. In ArtNet and sACN you can specify a "universe id/number". That number is used to decide on which pin the data will be sent/driven. Yes, memcpy is also used in the background but the firmware/code for all that is already done.

To go step by step: Do you have a pico-board connected via USB to your PC with the rp2040-dongle firmware running?

kripton, 2022-01-07:
Oh, and to avoid any mis-understandings: This project is not a library that you should include in you projects. It's a project for a "complete device" on its own. If you simply want to "use it", there's no code that you need to write. Just build/order the hardware and you're good to go.

kripton, 2022-01-08:
@ScobbyDoo : I've prepared a new version of the README: https://github.com/kripton/rp2040-dongle/tree/readme
Check it out and tell me if there's anything unclear :+1:

kripton, 2022-07-07:
For connrcting Ethernet, see #29. Closing this due to inactivity, feel free to reopen

closed 2022-07-07 by kripton