OpenLightingProject/rp2040-dmxsun #22: Implement RDM

kripton, 2022-01-07:
Lots of work to be done here :)
Depends on #21 

I personally don't have much experience with RDM. I do understand the concept and benefits. However, I don't have any real fixtures to test it, only the "Arduino RDM responder".

peternewman, 2022-01-10:
> Lots of work to be done here :) Depends on #21

You don't need the hard bit of #21 in break detection for RDM, but you will need a lot of other input stuff.

> I personally don't have much experience with RDM. I do understand the concept and benefits. However, I don't have any real fixtures to test it, only the "Arduino RDM responder".

The Arduino responder ( https://github.com/mathertel/DmxSerial2 ) should be fine, I've run the OLA tests against it and keep fixing issues. I can test with some proper gear too if I build myself a real version of this.

kripton, 2022-01-10:
Cool :+1: 
We should also re-check what parts of the JaRule-project can be re-used since that does RDM as well