OpenLightingProject/rp2040-dmxsun #32: Add some soft of source-code-level documentation

kripton, 2022-01-08:
It would be great to have some nice documentation about all classes and methods there are in the source code. I'm thinking of doxygen and the like but I'm open to other proposals as well.

peternewman, 2022-01-10:
Doxygen would certainly match up with the main OLA codebase, which is probably easier for people working between the two (e.g. when writing plugins). We could also use all our existing workflows for both publishing on the website and validation etc, rather than having to use other toolsets.

Unless there's a strong argument for something else I'd be inclined to go with it and could bolt most of that together, or are you expecting someone else to write the documentation too? :laughing:

kripton, 2022-01-10:
Sure, if there's already a working toolset, I agree that should be re-used and would be happy if would volunteer to set it up.

No, I'm not expecting that someone else writes the actual documentation, I can well do that. I'm not sure which level/amount would be good but I'll start and see when it's sufficient.

peternewman, 2022-01-10:
Well if you want to put a bit of Doxygen style comments in, that should be enough for me to get going on it.

kripton, 2022-01-10:
Haven't used doxygen much yet but I'll start adding some doxygen-style comments the following days

kripton, 2022-01-11:
First steps done, still a lot to document ;)
https://github.com/OpenLightingProject/rp2040-dongle/compare/main...kripton:sourceCodeDocumentation