OpenLightingProject/rp2040-dmxsun #39: Demo version of the website

peternewman, 2022-01-10:
It might be nice to push a demo version of the website up to GitHub pages or the Open Lighting website, assuming it could be made to work standalone or with some dummy JSON files without too much work.

kripton, 2022-01-10:
Indeed, good idea. Also, a thing that can well be done. For development, I host the web page on my laptop and the API calls go to the dongle. The data the API returns could also be random for the moment

kripton, 2022-01-11:
Okay, I had thought about a quite "dynamic" mock API that would generate some data randomly so the refresh (on the console page for example) are a bit more lively. However, that wouldn't be possible with GitHub pages (or any other static file hosting) and for a first look at how the Web UI works static data should be fine