OpenLightingProject/rp2040-dmxsun #40: Re-investigate on zeroconf / mDNS

kripton, 2022-01-11:
Currently, the device chooses one IP address based on the chip ID in the flash memory (if no IP has been stored in the config) and provides another IP in the same subnet to the host machine via DHCP.
This might be easier using AutoIP/APIPA provided by the lwIP stack: https://www.nongnu.org/lwip/2_0_x/group__autoip.html. We just need to make sure that all host OSes we want to support don't treat network connections as non-functional if their DHCP requests go unanswered.
Zeroconf is the name of the technology / processes ifself. Apple Bonjour and Avahi are two widely-known implementations.

Also, usbsun should at least advertise some kind of hostname and the services it provides via mDNS (also already supported by lwIP).

peternewman, 2022-01-12:
I didn't realise the mDNS bit and the standalone IP bit were both part of the same standard!

I think Zeroconf has pretty wide OS support now, although Windows may flag the network as no Internet access.

kripton, 2022-01-12:
Well, it's not necessarily. Sorry if I made that impression. Standalone IPv4 is in https://datatracker.ietf.org/doc/html/rfc3927, mDNS is in https://datatracker.ietf.org/doc/html/rfc6762 and https://datatracker.ietf.org/doc/html/rfc7558

The lwIP usually assumes you are using their AutoIP-feature when you activate their mDNS-feature but at least according to their docs, it's not a must:
`MDNS with IPv4 requires LWIP_IGMP = 1, and preferably LWIP_AUTOIP = 1.` (https://www.nongnu.org/lwip/2_0_x/group__mdns.html). IGMP is already in use for E1.31 multicast to work.

So while I'm still not sure about AutoIP, mDNS makes sense in any case.

And about Windows flagging the network as "No Internet access": That shouldn't be a problem since this is just the truth.