OpenLightingProject/rp2040-dmxsun #44: uf2-file

gobo-ws, 2022-06-03:
Hi,
I am struggling to understand GitHub actions but I guess I don't have enough patience at the moment :)

> If you just want to get started quickly find the most recent action run from the branch you're interested in, extract the attached artifact zip file (from https://github.com/OpenLightingProject/rp2040-dongle/actions/workflows/pico-build.yml ) and copy the rp2040-dongle.uf2 file to the Pico board when it's in bootloader mode.

I don't find any zip file or how to generate it?

kripton, 2022-06-04:
True, the results/artifacts are not easy to find if you are not used to it. Try this one: https://github.com/OpenLightingProject/rp2040-dmxsun/suites/6080805161/artifacts/210508458

gobo-ws, 2022-06-04:
Many thanks @kripton

closed 2022-06-04 by gobo-ws