OpenLightingProject/rp2040-dmxsun #49: Support Web interface, Art-Net, E1.31 and EDP via WiFi on the Pico W

kripton, 2022-07-06:
Raspberry Pi released the Pico W including on-board WiFi :partying_face: https://www.raspberrypi.com/news/raspberry-pi-pico-w-your-6-iot-platform/

While the radio-module doesn't support narrow-band RF in the 2.4GHz band and Bluetooth is not yet supported/enabled, the new Pico W is pin-compatible (except the on-board LED) with the current one.

It sounds like a good "step up" from the regular Pico and to enhance the dmxsun with the web interface, Art-Net, E1.31 and EDP via the WiFi-interface in addition to the emulated CDC NCM via USB.

Things needing work:
- The on-board-LED is now managed via the radio module. As such, GPIO 25 is no longer available and https://github.com/OpenLightingProject/rp2040-dmxsun/blob/main/src/pins.h#L45 leads to a compilation error. Easy fix
- pico-sdk 1.4.0 seems to ship lwIP in two places: `lib/tinyusb/lib/lwip/` (where it already was before) and `lib/lwip/` (where it was added to support IP via the cyw43 driver. This leads to the following compilation error (might be version conflict, needs investigation):
```
In file included from /opt/raspberrypi/pico-sdk/src/rp2_common/pico_cyw43_arch/cyw43_arch.c:12:
/opt/raspberrypi/pico-sdk/lib/cyw43-driver/src/cyw43.h:118:17: error: field 'dhcp_client' has incomplete type
  118 |     struct dhcp dhcp_client;
      |                 ^~~~~~~~~~~
```
- The cyw43 radio module seems to need two firmware blobs (https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/firmware/43439A0-7.95.49.00.combined, 220kByte and https://github.com/georgerobotics/cyw43-driver/blob/195dfcc10bb6f379e3dea45147590db2203d3c7b/firmware/4343WA1-7.45.98.50.combined, 416kByte). Compared to the 2MByte flash (which is the same since as with the regular Pico and has not been increased for the Pico W), those files are huge. The current main branch of rp2040-dmxsun generates a 1.5MByte uf2-file (debug build), mainly due to the React-based web interface. I'm afraid that with adding the two firmware blobs, the flash memory will not suffice :(. Not yet confirmed since I didn't yet fix the compilation error mentioned above

6by9, 2022-07-07:
```
In file included from /opt/raspberrypi/pico-sdk/src/rp2_common/pico_cyw43_arch/cyw43_arch.c:12:
/opt/raspberrypi/pico-sdk/lib/cyw43-driver/src/cyw43.h:118:17: error: field 'dhcp_client' has incomplete type
  118 |     struct dhcp dhcp_client;
      |                 ^~~~~~~~~~~
```

Most likely unrelated to 2 copies of lwip. 
https://github.com/OpenLightingProject/rp2040-dmxsun/blob/main/src/lwipopts.h#L45
```
#define LWIP_DHCP                       0
```
which means that "lwip/dhcp.h" is not included.

Next issue comes up of ```netif_list``` not defined, which is due to https://github.com/OpenLightingProject/rp2040-dmxsun/blob/main/src/lwipopts.h#L75
```
#define LWIP_SINGLE_NETIF               1
```

Link then fails as we have multiple definitions for ```sys_arch_protect```, ``sys_arch_unprotect``` and ```sys_now``` (pico-sdk/src/rp2_common/pico_lwip/nosys.c and rp2040-dmxsun/src/tusb_lwip_glue.c)

6by9, 2022-07-07:
Hacking done and it builds and runs for Pico-W. Branch pushed to https://github.com/6by9/rp2040-dmxsun/tree/pico-w

```
mkdir build
cd build
cmake -DCMAKE_BUILD_TYPE=Debug  -DPICO_BOARD=pico_w -DWIFI_SSID="Your Network" -DWIFI_PASSWORD="Your Password" ../src
make -j4
```

No code written to actually add the wifi side, but the USB network interface still works.
I probably should have based it off your https://github.com/kripton/rp2040-dongle/tree/ethernet branch, but there you go.

kripton, 2022-07-07:
Amazing, nice work! I can confirm that the compilation works and includes the cyw43-driver. However, I don't see the firmware getting included and the resulting uf2-file is still 1.5MByte.
Regarding flash memory consumption, here's the idea to compress the firmware: https://github.com/georgerobotics/cyw43-driver/issues/4

kripton, 2022-07-07:
Correcting my previous comment: That objcopy call wraps the binary firmware in a valid elf object file so it can be linked. Content/format of the input doesn't matter so precompresssed should be fine

kripton, 2023-02-08:
Mostly done with #68. WebUI, ArtNet and sACN work over WiFi when configured properly in the source code.