OpenLightingProject/rp2040-dmxsun #53: DHCP request not acked

cuvoodoo, 2022-08-16:
after the DHCP offer from the RP2040 over USB, the computer (USB host) requests the offered IP.
but the RP2040 does not ACK this request.
thus the DCHP negotiation does not complete and the IP is not assigned to the interface (at least nor under Linuc Arch using dhcpcd).

see attached trace where DHCP client repeats the unanswered request.
[image]

kripton, 2022-08-16:
Okay, that's strange, works on my machine. Will do a wireshark trace this evening an compare with yours.
In the meantime, is there anything strange in the kernel logs? USB device resetting or similar?

kripton, 2022-08-16:
Just checked, I can definitely see a DHCP ACK here:
[image]
Did you build the source code yourself or did you use some UF2-file from the GitHub actions?

The DHCP server is part of the lwIP-stack, it's not written by me. I might be able to add or active some debugging options but I didn't yet dig into it.

gobo-ws, 2022-08-17:
I had the same problem with a UF2-file from the GitHub actions (https://github.com/OpenLightingProject/rp2040-dmxsun/suites/6080805161/artifacts/210508458)

kripton, 2022-10-26:
Thanks for the pcap-file, that will help a lot. Unfortunately, I didn't yet find the time to investigate the root cause :/

kripton, 2022-11-01:
I did some investigation today. First thought: Only "a limited amount" of requests can be served. Proofed wrong.
Then I tried `dhcpcd` as DHCP client and I can reproduce the failing behaviour of not getting the `DHCP ACK`. However, when I use NetworkManager's internal DHCP client afterwards on the same connection (= without disconnecting the board), it serves the DHCP ACK just fine. However, still no explanation why this happens. The DHCP server code is not written by me but I will try to debug it to see where it fails.

kripton, 2022-11-01:
I think I found the root cause: tinyUSB's DHCP server expects the MESSAGETYPE option in DISCOVER or REQUEST packages to be the first option in the packet. Some clients seem to not to adhere to this expectations. The trace you attached and the "dhcpcd" I tried send the requested IP option first.

Pull-request for tinyUSB has been created, but it might take some time until the fix makes it into tinyUSB and from there into the pico-sdk. I might try to hack something together so the builds created by the GitHub actions behave correctly.

In the meantime, here's an UF2-file that includes the fix: 
[rp2040-dmxsun.zip](https://github.com/OpenLightingProject/rp2040-dmxsun/files/9914443/rp2040-dmxsun.zip). Might be a DEBUG build containing some more logging statements than required but it would be nice if you could confirm that this fixes the issue. Thanks!

kripton, 2022-11-18:
Fix has been merged upstream. I now need to check if I can update to the latest upstream master of tinyUSB or if there are incompatibilities that I need to fix. Thus, leaving this open

kripton, 2023-02-02:
Alright, I'm working on a way to fix this locally without waiting for the upstream version of tinyUSB to tickle down into the pico-sdk.

kripton, 2023-02-08:
Should be fixed with #68.
I copied the DHCP server code into the project to be able to modify it.

commit 70df7f52 (2023-02-08): Copy tinyUSB's DHCP server to our code base and use that instead so we can modify it. Plus make it multi-interface capable. Closes #53
closed 2023-02-08 by kripton