OpenLightingProject/rp2040-dmxsun #59: rp2040-dmxsun - system compilation error

trinityworld, 2022-11-17:
window system :-

I am getting complication error when run this command "make -j$(($(nproc)+1)) " and also window has issues finding libraries (lwip)

C:\Users\pico\Downloads\rp2040-dmxsun\pico-sdk\lib\tinyusb\lib\networking\dhserver.h:38:10: 
fatal error: lwip/err.h: No such file or directory

i cant find **.uf2** file in this repo. it will be helpful..

[image]

peternewman, 2022-11-17:
Have you installed the lwip library?

> i cant find **.uf2** file in this repo. it will be helpful..

See https://github.com/OpenLightingProject/rp2040-dmxsun#how-can-i-try-it-out-without-spending-much the artifacts for the run has them.

kripton, 2022-11-17:
The lwIP module is integrated as a submodule and it looks like you didn't clone them. Either run a 'git submodule update --init' or remove your working copy and run the clone again with the '--recursive' parameter

peternewman, 2022-11-17:
> Regarding the UF2 file from the actions: The problem is that they expire and they are only available to logged in users.

You could push them to GitHub pages a bit like I'm doing here if you wanted to offer bleeding edge ones easily for people:

https://github.com/mjemmeson/Date-Holidays-GB/pull/14/files

Although I guess anyone that wants bleeding edge probably has a GitHub account and can grab them...

> I hope GitHub keeps the artefacts when they are linked to a release. Which we don't have yet but will soon.

It does for normal release files, hopefully you can attach them in the same way then.

kripton, 2022-11-17:
Just for reference, latest UF2 is here: https://github.com/OpenLightingProject/rp2040-dmxsun/suites/9370700870/artifacts/441397266

trinityworld, 2022-11-18:
code run successfully and also get **.uf2 file** but webui not generate..

kripton, 2022-11-18:
Okay, that's already some good progress!
Yep, the way the Web-UI compilation is integrated is a bit of a mess (acknowledged as #61). Do you have NodeJS installed? If so, which version and is the "node" command in your %PATH%?

trinityworld, 2022-11-21:
yes, i installed Nodejs.
version is - v18.12.1 and also add "node" command in %PATH%.

[image]

kripton, 2022-11-29:
Hi, sorry, forgot to reply. Even though you seem to be using Ubuntu by now (#62), I would like to understand why this fails. Most probably since the Web UI is currently built by calling https://github.com/OpenLightingProject/rp2040-dmxsun/blob/main/build-webui.sh. And since there is no bash and some commands in that file will fail on the "Windows command line", this currently just doesn't work. Looks like a limitation of the current solution and tracked by #61.

trinityworld, 2022-11-30:
when i compiled this repo in Ubuntu then web ui generated.now got proper "uf2" file.

@kripton  Thank u so much..

kripton, 2022-11-30:
Glad to hear. You're welcome :) As I said, Linux is only currently supported build environment and it's cool that you managed in the VM!

kripton, 2024-01-05:
Closing as resolved

closed 2024-01-05 by kripton