OpenLightingProject/rp2040-dmxsun #62: pico rp2040  and Ethernet hardware configuration

trinityworld, 2022-11-29:
i am trying to Ethernet( ENC28J60 ) configuration with pico rp2040 as per https://github.com/OpenLightingProject/rp2040-dmxsun/compare/main...kripton:ethernet) but get error..

[image]

[image]

kripton, 2022-11-29:
Thanks for the report. In my repo, I can see that file located at `lib/lwip/contrib/apps/tcpecho_raw/tcpecho_raw.h`. Can you confirm this in your local copy?
If not, can you please cd into the directory `lib/lwip`, run `git describe` and show us its output? For me it is showing `STABLE-2_1_0_RELEASE-522-g88ce4385`

trinityworld, 2022-11-30:
> Besides, do you have any specific reason to use the ENC28J60? I'm currently re-designing the PCBs and I would switch to the WZ5500 chip since there are modules available which are cheaper and easier to integrate. If there are no objections or things that the ENC28J60 can do better.

- No reason , just want to drive with Ethernet & this repository already working on it
-  W5500 > yes, i also recommend this module (they have internal ip stack )
-  If you have any documents with w5500(schematic,workflow,firmware) i will work on that & add pull request
-  should i try this board (as 6 to 21 Gpio is used for 16 parallel output) for SPI need bitbang or some other trick 

[image]

**- if not > what will pin connection with Ethernet module (0 to 5 Gpio)**

trinityworld, 2022-11-30:
[image]

i remove this line " #include "tcpecho_raw.h" from "lwip_integration.c" file as per above mention repository then my previous error  solved.
[image]

but here,
lwip/../../http/fs.c

library error occurred.

[image]

trinityworld, 2022-11-30:
why not add  this ->> "lwip_integration.c" file in add_executable list CMakeLists.txt ?..
 
[image]

kripton, 2022-11-30:
>     * No reason , just want to drive with Ethernet & this repository already working on it

Well, yes. The ` ethernet` branch is really work-in-progress, it's nothing production-ready. Also the Pico got stuck whan a lot of traffic was created (Denial-of-service :()

>     * W5500 > yes, i also recommend this module (they have internal ip stack )

I know that they have an internal stack but I don't plan to use it. I'd rather keep the software generic to not depend on a specific chip model

>     * If you have any documents with w5500(schematic,workflow,firmware) i will work on that & add pull request

The W5500 (the WIZ850io module to be exact) is added in this branch: https://github.com/kripton/rp2040-dmxsun/tree/kicadUpdates. The baseboard_4slots has a socket for the module and the IRQ line is shared with the nRF24 module. For the new baseboard_2lots, there is a jumper and configurable IRQ pins since less are used for the DMX output. PDFs of schematic and board are available via the actions: https://github.com/kripton/rp2040-dmxsun/actions/runs/3578552144

>     * should i try this board (as 6 to 21 Gpio is used for 16 parallel output) for SPI need bitbang or some other trick

Yes, 6 to 21 are used for DMX output. However, SPI is used to communicate with the nRF24 and I shared this SPI with the WIZ850io. Thus, only one additional CHIP-SELECT pin is required (and the former debug trigger PIN is used as shared IRQ).

> **- if not > what will pin connection with Ethernet module (0 to 5 Gpio)**

Yes, I've seen that module. Unfortunately, the pinout seems to be incompatible with what I've chosen for the dmxsun

kripton, 2022-11-30:
> i remove this line " #include "tcpecho_raw.h" from "lwip_integration.c" file as per above mention repository then my previous error solved
> but here, lwip/../../http/fs.c
> 
> library error occurred.

It all looks to me like the lwip submodule doesn't match what the other files are expecting. Please see my request from above to show us the output of `git describe` in the `lib/lwip` directory. You can also try running `git submodule update` after you checked out the `ethernet` branch. This should bring the lwip submodule to the correct revision.
In the meantime, I will try with a fresh clone in a clean environment.

kripton, 2022-11-30:
> why not add this ->> "lwip_integration.c" file in add_executable list CMakeLists.txt ?..

Because usually, one links in / references the CMakeLists file from all libraries and the libraries should add the files to be compiled by themselves.

I've just tried compiling this from scratch. On my RPi4 it fails, but with another error and the reason is that gcc 8.3 is the latest version available and that doesn't support C++20, which is required for `src/webserver.cpp`. On my development machine, the `ethernet` branch still compiles fine and creates the .uf2-file.

trinityworld, 2022-12-01:
> It all looks to me like the lwip submodule doesn't match what the other files are expecting. Please see my request from above to show us the output of `git describe` in the `lib/lwip` directory. You [...]

[image]

trinityworld, 2022-12-02:
can you please check where is my mistake for assign spi and INT_PIN PIN..

1>lib/pico-enc28j60/enc28j60_lwip_glue.c

[image]

2> lib/pico-enc28j60/enc28j60_lwip_glue.h
[image]

3>src/lwip_integration.c

[image]

4>src/pin.h

[image]

5>src/main.c

[image]
[image]

trinityworld, 2022-12-05:
here, i get .uf2 file but ..

[image]

when i ping this static IP address.

[image]

"Destination Host Unreachable"  error generate

[image]

trinityworld, 2022-12-19:
pico RP2040 Ethernet branch build successfully..

kripton, 2022-12-19:
Oh nice, tbis is good to hear. Do you mind sharing some insights whaf fixed it for you?

kripton, 2022-12-28:
@trinityworld do you want to tell us what was wrong or how you fixed it? Otherwise I would close this issue.

trinityworld, 2023-01-06:
GPIO 27 use for ENC-INT pin then problem solved..

now i am build project only get 16 parallel output.

https://github.com/trinityworld/pico_dmx/blob/b34064bc0b26339a80139178447e03a13bfb78d1/src/main.c#L102

trying to fill wavetable buffer from 16 universe data (buffer)and after than buffer transfer to 16 gpio as parallel.but here  :(  not get dmx output.

kripton, 2023-01-06:
Okay, I never used the ENC's INT pin on the dmxsun

And I'm getting a HTTP 404 when I try to access your link

trinityworld, 2023-01-09:
https://github.com/trinityworld/pico_dmx/blob/b34064bc0b26339a80139178447e03a13bfb78d1/src/main.c#L102

kripton, 2024-01-05:
@trinityworld : I'd propose to close this issue since it's mainly about the ENC28J60 which will not be used for the dmxsun. I will rather go with the W5500. If you want to keep the support in your fork and it doesn't conflict with other parts, I will gladly accept a PR.