OpenLightingProject/rp2040-dmxsun #64: pico only blinks

jazzvivi2, 2022-12-21:
Hello, I'm trying to test your project but I load the latest version of your software and my pico only flashes the led quickly but no network device appears. load some sample uf2 projects and it works. im very new with pico am i forgetting something?

thank you!

kripton, 2022-12-21:
Hi @jazzvivi2 : It depends on how fast the LED blinks. If it blinks VERY FAST (~5-10 times per second), it hasn't yet finished initialization and won't do anything useful. If it blinks slower (~1-2 times per second), it should enumerate as USB device and should be ready to use.

Did you compile the UF2 yourself or did you use one from the GitHub actions?
Which host system (OS) are you using to connect your Pico?
Any hardware attached to the Pico board?

jazzvivi2, 2022-12-21:
Hi @kripton Thanks for answering, the blinking is very fast, the led hardly has time to turn off. I'm using GitHub actions file I did tests on windows and osx and I don't have anything connected to my pico board

kripton, 2022-12-21:
Can you give me the link to the action you took the UF2 file from? I will try to run that here on my Picos. The only thing that I can imagine right now is that your Pico is not able to handle the frequency the firmware configures the cores to: https://github.com/OpenLightingProject/rp2040-dmxsun/blob/main/src/main.cpp#L99. I can build you an UF2 tomorrow that doesn't overclock. Or at least to a lower frequency.

jazzvivi2, 2022-12-21:
https://github.com/OpenLightingProject/rp2040-dmxsun/suites/9865571969/artifacts/475895047

Thanks for everything

kripton, 2022-12-21:
I've tested that UF2 file on three of my Picos. They all behave "good" (= LED blinks very fast for about a second but then changes to the slower blinking pattern).

Here's the variant without overclocking (as soon as the build finishes): https://github.com/kripton/rp2040-dmxsun/actions/runs/3753562092

Do you have another Pico that you could try? Maybe the one that you have was a bit unlucky regarding the process parameters and the cores cannot be overclocked that much?

jazzvivi2, 2022-12-21:
Yes now!! You were right, the problem is speed.
my motherboard is a chinese version (purple with 16mb)
https://es.aliexpress.com/item/1005003260584517.html?spm=a2g0o.order_list.order_list_main.11.2201194dHxQOz1&gatewayAdapt=glo2eng

I'm going to play with this and I'll tell you
thank you very much again

kripton, 2022-12-21:
Oh, cool that it worked!
If it's from aliexpress, it's most probably not a "Pico", at least not an original one ;)
And I assume it's the flash that is not able to keep up with the faster clocks. If the SYS clock is made faste, this affects the flash clock as well and in the youtube-video mentioned in main.c, it's also the flash that quits at some point. One could modify the 2nd stage bootloader to clock the flash slower and have the cores still at 250 MHz but probably not worth the effort.
At least good to know that some 16MB flashes cannot do with sysclk at 250MHz!

kripton, 2022-12-21:
It's also interesting that the pin-layout on the edges is different compared to the original Pico

jazzvivi2, 2022-12-21:
you know... aliexpress is always fake. sometimes they improve the designs and many others annoy them. tomorrow I will try to connect an rs485 module and a w5500

kripton, 2022-12-22:
Nice :) You know that the code to get the W5500 working has not even been started? ;)

kripton, 2022-12-22:
Closing this since we found the root cause and workaround

closed 2022-12-22 by kripton

jazzvivi2, 2022-12-26:
hello kripton, first of all congratulations on the holidays. to get dmx output am i supposed to just connect a tll-rs485 module? not having an eprom the board is not recognized. the led of my rs485 board stays on and does not blink like when I receive data

kripton, 2022-12-27:
Indeed, at the moment the EEPROMs don't matter much, the DMX signals generated anyways. Hooking up an RS-485 driver should be enough. Make sure you have the DE and !RE wired properly.
Which LED do you mean?

jazzvivi2, 2022-12-27:
Hello @kripton !

I normally use these boards

https://es.aliexpress.com/item/1005001621746811.html?spm=a2g0o.productlist.0.0.59ca339eeP2AEW&ad_pvid=2022122707553412505464808256960004470222_1&s=p

they do not have a "de" or "re" pin
Both the input and the output are activated.

If you look closely, this board includes two LEDs, these blink when they receive or send data, in my case it stays on permanently when I connect gpio. I tried all of them and I can't get an output. in the console if I can see the values ​​that it sends for example qlc+

kripton, 2022-12-28:
very hard to diagnose without access to your hardware. Having the LED constantly lit is not necessarily a bad thing, it could be blinking so fast that your eyes csnnot see that it's actually blinking.
Do you something like an oscilloscope or logic analyzer available that you can connect to the GPIOs to see what they are outputting?