OpenLightingProject/rp2040-dmxsun #77: No new .uf2 build last 5 months

ultraw3ird, 2023-08-15:
Currently there is no way to get compiled .uf2 file from this Github. Would it be possible to check bot for automatic builds as all artifacts are long expired? It is pain in the *** to compile it on Windows and not all guys has dedicated linux machine for such a purporse. Thanks in advance

peternewman, 2023-08-15:
Hi @ultraw3ird 

I've opened #78 which should also kick off it's own build.

The artefact should be here:
https://github.com/OpenLightingProject/rp2040-dmxsun/actions/runs/5867409629

@kripton we should really publish nightly builds to a GitHub Pages website or something too...

peternewman, 2023-08-15:
Doing formal releases with attached builds would make sense too and could probably be done mostly automagically:
https://github.com/OpenLightingProject/rp2040-dmxsun/releases

kripton, 2023-08-16:
@ultraw3ird : Sorry that no UF2 was available since there was not much progress on this project. Also thanks for the praise that it's nice HW and SW. Did you order & assemble the hardware?
I agree that having an official release would make sense. It's just that I have too much on the ToDo that I'd like to have in the "first release" and I'm pretty bad at "descoping" features so that we could we release with what we currently have ... :/

PR #78 (2023-08-16): Also run pico-build.yml weekly
Closes #77
closed 2023-08-16 by kripton